Ecam-Eurobot/Eurobot-2017 #13: UML Raspberry

JonathanPetit, 2017-02-22:
Uml complet du projet eurobot 2017. Réfléchis par Charles et moi-même.
Suggestions ou corrections dites le nous.
@azerupi Qu'en penses-tu?

[image]

azerupi, 2017-02-23:
En reprenant les classes que nous avons déjà pour les modules I2C, personellement je commencerais par étendre comme ceci:

[image]

L'idée c'est d'avoir une classe `Robot` qui contrôle le robot à un haut niveau (éventuellement cette classe pourrait aussi avoir la routine principale, e.g. `run`). Cette classe contiendrait les différents modules qu'on à implémenté (capteurs, moteurs, pinces, ...) et fournit les méthodes de plus haut niveau.

Il contiendrait également une classe `Map` qui gère la position des différents obstacles et le pathfinding. Je pense qu'un map en tant qu'array 2D va être beaucoup trop coûteux. Il y à moyen de fortement s'inspirer des algorithmes et représentations utilisés pour les jeux. J'avais pensé à travailler simplement avec des listes de coordonnées cartésiennes et des calculs de "bounding box" pour les obstacles, mais il faut voir si ça fonctionne pour le pathfinding. Je pense qu'on peut ouvrir une issue rien que pour discuter de ça :yum:

charlesvdv, 2017-02-24:
En reparlant avec M. Marchand, les différentes stratégies n'ont pas d'interets si on a un algorithme de recherche de chemin performant pour faire volte-face si on a un obstacle. Du coup je suis d'accord que ca n'a pas d'intéret d'avoir des classes ```HighLevelAction``` et ```Strategy``` et mettre tout dans la classe ```Robot```.

Concernant la map, j'ai trouvé un truc vraiment pas mal... C's un [navigation meshes](http://theory.stanford.edu/~amitp/GameProgramming/MapRepresentations.html#navigation-meshes) qui représente les chemins possibles par un graphe. Ca a l'air d'être la solution la plus optimal. 

On pourrait également avoir des meshes avec différentes zones pour par exemple limiter la vitesse(bascule), etc...

Je pense également qu'il serait intéressant d'avoir des nodes spécifiques comme atouts où il pourrait avoir des contraintes d'angles pour arriver sur l'objet, etc... et de laisser les obstacles abstraits (sans avoir de types précis) pour avoir plus de flexibilité pour les années d'après.